ros/common_msgs #58: Acceleration Message

paulbovbel, 2015-03-20:
I was debating the idea of adding an 'acceleration' type message to complement twist, pose, wrench. This would be very helpful for some controller work where accelerations are passed around between several components.

While trying to come up with an appropriate name, I'm guessing that geometry_msgs naming conventions (Twist, Wrench) came out of the KDL classes? http://people.mech.kuleuven.be/~rsmits/kdl/api/html/annotated.html

KDL represents both acceleration and velocity information using the Twist semantic (TwistVel and TwistAcc). What would be the preferable option:
1) Update the description of 'Twist' messages to include velocity AND acceleration
or
2) Add Accel.msg, AccelStamped.msg, etc. to provide clear separation between the two types
or
3) Do nothing at all

paulbovbel, 2015-03-20:
I would be happy to do a pull of either-or.

tfoote, 2015-03-20:
Definitely not 1. Velocity and Acceleration are semantically different. 
2 seems reasonable. Twist is pretty standard for naming velocity. The datatype for acceleration has the same layout but does not have a canonical name. I think Accel and AccelStamped would be reasonable. 

It would be great if you could provide converters for kdl's TwistAcc to make it useful out of the box too. https://github.com/ros/geometry_experimental/blob/indigo-devel/tf2_kdl/include/tf2_kdl/tf2_kdl.h

tfoote, 2015-03-22:
#59 #60 into indigo and jade

closed 2015-03-22 by tfoote