ros/common_msgs #62: geometry_msgs/InertiaStamped.msg depends on non-existent package telemanip_msgs

airballking, 2015-04-13:
I think we found a bug in geometry_msgs (branches indigo-devel and jade-devel):

The new message definition ```geometry_msgs/InertiaStamped.msg``` depends on the non-existent package ```telemanip_msgs```. 

This was introduced by this pull request https://github.com/ros/common_msgs/pull/55

Can somebody confirm this issue?

airballking, 2015-04-13:
I provided a fix for Indigo. It works under the assumption that Jonathan actually wanted to use ```Inertia.msg``` which he also pull requested into ```geometry_msgs```. @jbohren Is this correct?

jbohren, 2015-04-13:
Yeah that's correct. Bad migration from the other package it was living in.

tfoote, 2015-04-17:
Fixed in #63

closed 2015-04-17 by tfoote