ros/common_msgs #84: return may reach end of non-void function

defunctzombie, 2016-04-26:
When building with `-Wall Werror` on the latest version of indigo I am hit with the following error during compilation:

```shell
/opt/ros/indigo/include/sensor_msgs/point_field_conversion.h:123:5: error: control may reach end of non-void function [-Werror,-Wreturn-type]
    }
```

**Compiler**:
clang 3.6

**OS**:
Ubuntu 14.04

This prevents my project from using `-Wall -Werror`.

I can make a patch but it is unclear to me how you would prefer the error condition be handled here.

tfoote, 2016-04-26:
Indeed the "error" case should not be hit as long as we have full coverage of the valid datatypes. Maybe an assert followed by a return of the default value to make the compiler happy.

defunctzombie, 2016-04-26:
@tfoote what would you consider the default value to be?

And for assert, would I use the builtin system asserts or is there a more "ROS" like way that is preferred?

pumaking, 2017-10-23:
Hitting this on kinetic as well.

tfoote, 2017-10-24:
In this case a ROS_BREAK would be recommended: http://wiki.ros.org/CppStyleGuide#Assertions

You can just make a T and return it with it's default value I think since we should never get there it won't matter the value.

pumaking, 2017-10-24:
Would adding a dependency on rosconsole be needed / all right?

jasonimercer, 2018-01-30:
Any thoughts on throwing a `std::invalid_argument` exception? This gets interpreted as a correct return path (compiler doesn't warn), has more semantic value than an abort and doesn't require adding a dummy return value.

tfoote, 2018-01-30:
That would also be a valid approach. We couldn't add that in mid release though. And we'd have to make sure to document the possible exception at each of the levels of API above this.

tfoote, 2018-02-12:
Warning addressed in #118

closed 2018-02-12 by tfoote