ros/common_msgs #89: Support for additional distortion models

reaganlo, 2016-07-25:
Based on this file https://github.com/ros/common_msgs/blob/indigo-devel/sensor_msgs/include/sensor_msgs/distortion_models.h, the supported distortion models are "plumb_bob" and "rational_polynomial"

The distortion models in two of our cameras use "Modified Brown-Conrady" and "Inverse Brown-Conrady". 

- Are there any plans to support these distortion models too? 
- If no, are we not supposed to create "sensor_msgs/CameraInfo" ROS messages for these unsupported distortion models?

tfoote, 2016-07-26:
We'd be happy to take contributions to add support for these models, though I think these basically map to the models that [OpenCV uses](http://opencv-users.1802565.n2.nabble.com/OpenCV-2-2-New-Rational-Distortion-Model-td5807334.html). [related](https://stackoverflow.com/questions/9465525/camera-calibration-rational-distortion-model) It might be easier to find the closest approximation based on the rational_polynomial distortion coefficients.

@vrabaud are you familiar with the different distortion models and their accuracy?

vrabaud, 2016-07-27:
Any model is supported, that's why the model is a string and not an enum https://github.com/ros/common_msgs/blob/indigo-devel/sensor_msgs/msg/CameraInfo.msg#L60
Explained here: http://www.ros.org/reps/rep-0104.html#alternate-distortion-models
Now, it depends on which pipelines you use next and those are the ones who need to adapt to your model.

reaganlo, 2016-08-24:
@vrabaud Thanks

closed 2016-08-24 by reaganlo