ros/common_msgs #113: NONE center aligned laser beam?

BigBro2014, 2017-09-22:
As cheap LIDARs use triangular method to mesure distance, and the laser beam is not center aligned, the angle is not accurate.
I suggest we add "x offset" to laser scan msg, so the gmapping_slam can get accurate reflection points.

laser beam
*
│
│
│
└─────(rotation center)

Does anynone have better ideas?

tfoote, 2017-09-27:
Adding this field would be pretty disruptive for a lot of existing laser driver and bag files and adds complexity to the projections algorithms. If we were to make that change all of these things would need to be updated simlutaneously.

There are several current workarounds. One is to resample the points. Another is to simply ignore that extra geometry. Many of the triangulation based lasers do not have accuracy/precision smaller than the x offset except for very short range.

The most common solution is to have the driver go directly to point clouds instead of using the laser message such that it can deal with any of the unique geometry elements of that sensor directly.

BigBro2014, 2017-10-26:
Point cloud is a good way. Thanks @tfoote

closed 2017-10-26 by BigBro2014