ros/common_msgs #116: Pproposing sensor message for DVL

narcispr, 2017-12-15:
Hi,
The underwater robotics community working with ROS is growing every day. Despite most of the sensors that we use are similar to other robotic domains (IMU, GPS, range, pressure, ...) there is a basic sensor, the Doppler Velocity Logger, that is not available in the sensor_msgs package. This sensor measures velocities wrt. the sea bottom and/or a mass of water. Additionally it can measure the range wrt. to the bottom, temperature, ... 

Right now there are several public messages for this sensor. 
* Some are based on one of the most popular DVL devices (Teledyne Explorer DVL):  https://github.com/uji-ros-pkg/underwater_simulation/blob/kinetic-devel/underwater_sensor_msgs/msg/DVL.msg
* Others try to be more generic: https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_sensor_plugins/uuv_sensor_plugins_ros_msgs/msg/DVL.msg

I think people developing the uuv_simulator (@sebastianscherer, @musamarcusso), the auv_msgs package (@IgnacioCarlucho, @bmagyar) or many other people in the underwater community could be interested on having a standard message for the DVL in the sensor_msgs package. 

How can we discuss about this topic?
Thank you!

tfoote, 2017-12-15:
You're welcome to use this thread to discuss. Or alternatively open a Pull Request with a proposed new message which would be fine. Though it looks like there's some pretty big difference between the two current implementations.

There probably needs to be some discussion/back and forth about what to capture atomically and can be generic vs sensor/vendor specific.

Since there are a few underwater vehicle projects and it's likely to be some pretty specific conversations I might suggest proposing an Underwater Vehicles category on discourse. Here's the process for a migration https://discourse.ros.org/t/ros-sig-migration-method/35 which we can also use for creation. That might be a good place to discus this and other underwater specific topics in the future. I'd like to see a few people commit to using the category. Otherwise the other recommendation I would have would be a thread in the [ROS Projects](https://discourse.ros.org/c/ros-projects) category as a intermediate place for a specific scoped proposal.

bmagyar, 2017-12-18:
The way I see it for `auv_msgs` to exist as-is is that the underwater community is not that big within robotics. This allows to make changes more dynamically, sometimes even breaking backward compatibility. We can get to common terms through short discussions.  We are happy to host a generalised version of the DVL message in `auv_msgs` for the time being.

narcispr, 2017-12-18:
I think it will be better for a standard DVL message to be inside the sensor_msgs package as it is a message for a sensor :) 
I'm sure that all the groups working with underwater vehicles are using sensor_msgs package but probably most of them are not using the auv_msgs one.
Using the definition proposed by the uuv_simulator people seems good to me. 

https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_sensor_plugins/uuv_sensor_plugins_ros_msgs/msg/DVL.msg

std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
geometry_msgs/Vector3 velocity
  float64 x
  float64 y
  float64 z
float64[9] velocity_covariance
uuv_sensor_plugins_ros_msgs/DVLBeam[] beams
  float64 range
  float64 range_covariance
  float64 velocity
  float64 velocity_covariance
  float64 yaw
  float64 pitch

The only thing that it is not clear in this definition is if the velocity is measured wrt the sea bottom o wrt a mass of water (there are DVLs that can swap from one to another or give both for each measurement). However, for these devices an easy solution could be to publish two topics: one for bottom tracking and another for water tracking measures.

@IgnacioCarlucho and @bmagyar, what do you thing about moving your DVL msg from uuv_sensor_plugins_ros_msgs to sensor_msgs?

This will allow people to have the same bag files either simulating an AUV or executing the same mission with a real vehicle.

Thank you!

tfoote, 2017-12-20:
FYI: There's a proposal for a new maritime robotics SIG: https://discourse.ros.org/t/maritime-robotics-sig-proposal/3489/1

narcispr, 2018-10-18:
Hi again!
After some discussion about this topic at [discourse.org](https://discourse.ros.org/t/common-message-structures-for-uuvs/3737/16) we organized a workshop at BTS to see which standard sensor messages are missing for the marine community. You can see some conclusions [here](https://discourse.ros.org/t/bts-workshop-adoption-of-conventions-in-the-underwater-ros-community-a-summary/6478). We agree on the definition of several messages. For this purpose a [repository](https://github.com/udgcirs/marine_ros_conventions_discussion) was created to foster discussion. For now, the one that is more advanced is the [DVL](https://github.com/udgcirs/marine_ros_conventions_discussion/blob/master/sensor_msgs/DVL.msg). I've forked both `common_msgs` and `rviz`  repositories and added the proposed DVL and DVLBeam mesages as well as an rviz visualization plugin. How should I proceed? Thank you!

closed 2025-05-31 by sloretz