ros/common_msgs #123: MultiDOFJointTrajectoryPoint message does not make sense

adamconkey, 2018-03-12:
By its name, the MultiDOFJointTrajectoryPoint should be encoding joint positions, velocities, and accelerations for an n-DOF manipulator. But it is using Cartesian space messages (Transform and Twist). This seems like it was a hack to get multi-DOF and can only go up to 6-DOF.

This message should probably not exist, MultiDOFJointTrajectory should accumulate JointTrajectory msgs along with the list of joint names, and there should be a CartesianTrajectory message, along with CartesianTrajectoryPoint, the latter having poses, twists, and wrenches.

adamconkey, 2018-03-12:
Actually, I notice now that JointTrajectory is exactly a multi-DOF trajectory, it has joint names and the ability to include multiple JointTrajectoryPoints, each of which specifies a trajectory for one DOF. So I don't know what MultiDOFJointTrajectory(Point) is trying to achieve.

tfoote, 2018-03-12:
This is specifically different from the JointTrajectoryPoint in that is supports MultiDOF Joints that cannot be represented by single DOF representations. In particular it's often used for expressing freespace trajectories.

closed 2018-03-12 by tfoote

adamconkey, 2018-03-12:
Are you saying then that this is for single joints, with multiple degrees of freedom? I'm not sure I understand the use case. It still seems like this package lacks support for Cartesian trajectories where I want a list of poses, twists, and wrenches.

tfoote, 2018-03-12:
The MultiDOFJointTrajectory specifically supports cartesian trajectories with poses, velocities, and accelerations expressed per "joint" as a time series. If you have more questions please use answers.ros.org following our [Support Guidelines](http://wiki.ros.org/Support)

adamconkey, 2018-03-13:
Fair enough, I think it's misleading to put Joint in the message name though when it's not in joint space.