ros/common_msgs #165: NavSatStatus quality status

droter, 2020-08-14:
Would there be any interest in extending the NavSatStatus message to include a quality status?  Right now it isn't easy to see what type of position quality you have.

Proposed extension:
---
# Navigation Satellite fix status for any Global Navigation Satellite System

# Whether to output an augmented fix is determined by both the fix
# type and the last time differential corrections were received.  A
# fix is valid when status >= STATUS_FIX.

int8 STATUS_NO_FIX =  -1        # unable to fix position
int8 STATUS_FIX =      0        # unaugmented fix
int8 STATUS_SBAS_FIX = 1        # with satellite-based augmentation
int8 STATUS_GBAS_FIX = 2        # with ground-based augmentation

int8 status

# Bits defining which Global Navigation Satellite System signals were
# used by the receiver.

uint16 SERVICE_GPS =     1
uint16 SERVICE_GLONASS = 2
uint16 SERVICE_COMPASS = 4      # includes BeiDou.
uint16 SERVICE_GALILEO = 8

uint16 service

**# Define GPS quality from NMEA-183 GGA Standard

int8 STATUS_NOT_VALID =  0        # Fix not valid
int8 STATUS_FIX =      1                 # GPS fix
int8 STATUS_DGPS_FIX = 2          # Differential GPS fix, OmniSTAR VBS
int8 STATUS_RTK_FIX = 4             #  Real-Time Kinematic, fixed integers
int8 STATUS_FLOAT_FIX = 5        # Real-Time Kinematic, float integers, OmniSTAR XP/HP or Location RTK

int8 quality**
---

NMEA-183 GGA Sentence https://www.trimble.com/OEM_ReceiverHelp/V4.44/en/NMEA-0183messages_GGA.html

nxdefiant, 2020-09-18:
Agreed I would like to see STATUS_RTK_FIX and STATUS_FLOAT_FIX (STATUS_FLT_FIX) as well.

gustavovelascoh, 2021-04-27:
+1 on including RTK fix and float status.

RonaldEnsing, 2021-07-02:
I would also be in favor of being able to distinguish between Differential, RTK-Float and RTK-Fixed status. The current `NavSatStatus` does not provide enough level of detail.

rsasaki0109, 2022-03-29:
@hidmic @tfoote @dirk-thomas 
I am very much in favor of this extension, but how can this proposal be incorporated into this package?
Since ros1 is eol in 2025, is there any more changes to be made?

tfoote, 2022-04-08:
To push this forward please make a proposal on the new repo for ROS 2 https://github.com/ros2/common_interfaces We can target it into rolling and then it will get into future distributions. 

Over there we'll want to see a discussion of what you'd like to change with examples of how it will be used and showing the improved performance. As well as consideration for how this will effect current users and any migration requirements necessary.

closed 2022-04-08 by tfoote

peci1, 2023-11-14:
The related discussion is here: https://github.com/ros2/common_interfaces/pull/220 .