ros/common_msgs #174: TransformStamped and tf2 Inconsistency

aock, 2021-06-28:
Since I want to transform some points from the robot's base into the map space, I used the tf2 function lookupTransform with a target frame "map" (1st argument) and source frame "base" (2nd argument). Accordinging to the documentation of TransformStamped.msg

```
# This expresses a transform from coordinate frame header.frame_id
# to the coordinate frame child_frame_id
...
```

I assumed tf2 would fill the message like

```cpp
TransformStamped T;
T.header.frame_id = "base";
T.child_frame_id = "map";
```

but it doesn't. The resulting message contains flipped child_frame_id and header.frame_id. So based on my assumptions, either the tf2 documentation or the TransformStamped documentation must be wrong. Or am I missing something here?

Thanks in advance,
Alex

hidmic, 2021-06-28:
Normally I'd suggest you make this question in https://answers.ros.org/, but in reflection, the phrasing _is_ confusing. 

IIUC, and @tfoote please correct me if I'm wrong, the `Transform` messages conventionally express the rotation and translation that the base frame _axes_ would undergo to reach the child frame _axes_. That is the opposite of transforming coordinates defined in the base frame to coordinates in the child frame, which is why you get the flip. In your example, applying some:

```
TransformStamped T;
T.header.frame_id = "map";
T.child_frame_id = "base";
```

to a coordinate in the "base" frame will transform it to the "map" frame, which is what you get when calling `lookupTransform()`.

tfoote, 2021-07-08:
That is correct @hidmic 

It's a common confusion and is why lookupTransform uses target and source frame which is explicity different than parent and child terminology for the transform trees themselves. And if you use the built in functions such as `transformPoint` you never even see the underlying transform.

closed 2021-07-08 by tfoote