ros/common_msgs #175: Redundant messages between Transform and Pose

ruelj2, 2022-02-10:
[Transform](https://github.com/ros/common_msgs/blob/noetic-devel/geometry_msgs/msg/Transform.msg) and [Pose](https://github.com/ros/common_msgs/blob/noetic-devel/geometry_msgs/msg/Pose.msg) are semantically the same thing. It is redundant to have both of them. I suggest keeping only Pose and its related stuff.

peci1, 2023-04-09:
In my view, they are not **semantically** the same. They might be mathematically the same (or very similar), but the semantics differ.

The general policy in ROS is to have separate message definition if they carry a semantically different information (even if the data structure would be the same). See e.g. `Point` vs. `Vector3` types.

Pose is (from msg definition) "pose in free space". It does not specify the pose of what kind of object.

Transform is "transform between two coordinate frames". Here it does specify that you should only apply it to coordinate frames.

The difference is nicely visible in the stamped variants of the messages. `TransformStamped` has the `child_frame_id` field, because it knows it specifies the tf between two tf frames. `PoseStamped` has nothing like that - it is just a generic pose of "something" in the given frame.

tfoote, 2023-11-21:
Yes they are differentiated due to the semantic differences highlighed by @peci1 above. 

Also there was a long thread on it recently at https://github.com/ros2/common_interfaces/issues/152

closed 2023-11-21 by tfoote