ros/common_msgs #181: Default Quaternion is invalid

Vini-002, 2022-08-03:
I just tried to use the default quaternion - `Quaternion()` - in a situation where I don't have rotation information and got an error that this quaternion is invalid. The solution was really simple, just added `w = 1` as a keyword parameter. I would suggest that to be the default quaternion.

I'm new to open source... but I imagine I should fork, change that and submit a pull request, right?

Vini-002, 2022-08-03:
I've just discovered the code I was looking at was automatically generated by genpy and don't know if it is possible to modify it.

peci1, 2023-04-09:
I don't think such PR would go through. Zero-initialized fields are a pretty standardized thing in ROS messages and nobody should expect anything else. It's the same with colors: if you set just RGB, and forget to set alpha to 1.0, your markers do not show up. The takeaway is - you always need to know and understand the data you are sending.

closed 2025-05-31 by sloretz